nix-community/nixos-facter #62: Review how we detect gpu

brianmcgee, 2024-08-01:
I think we are currently capturing it, but check how we are. For example I see my own graphics card coming up as a `display` and it's saying "unknown graphics card hardware"

brianmcgee, 2024-08-09:
`hwinfo` pulls ids from various public sources and has a script for updating them: https://github.com/openSUSE/hwinfo/tree/master/src/ids

It looks like `usb` and `pci` ids were last updated in 2021. After running the update script I could see my graphics card listed. 

I have created a PR upstream: https://github.com/openSUSE/hwinfo/pull/144 and will testing with my fork https://github.com/brianmcgee/hwinfo

brianmcgee, 2024-08-09:
@Mic92 see above

brianmcgee, 2024-08-09:
It appears when I introduced the `--hardware-features` flag to reduce the default list of things that `hwinfo` probes, I introduced a bug with how we resolve model names and so on. If you run `nixos-facter -f all -o report.json` you will see that we resolve model names and other lookups fine. But with the default probe list `nixos-facter -o report.json` you start to see a lot of unknown hardware entries in the report. 

I'm reviewing how `hwinfo` uses the `lib` to identify how we should be using it better. For now, running it with `-f all` seems to resolve the issue.

@Mic92 this could also explain why you weren't seeing a lot of things with their model names and hardware classes properly resolved

Mic92, 2024-08-09:
Okay

PR #72 (2024-08-13): fix: lookup of pci and usb ids when specifying hardware features
We were losing pci and usb lookups when using anything other than `all` for `hardware-features`.

This fixes that by mixing in the `int` probe feature regardless of the list provided, matching this behaviour https://github.com/openSUSE/hwinfo/blob/c87f449f1d4882c71b0a1e6dc80638224a5baeed/src/hd/hd.c#L597-L605

Closes #62
closed 2024-08-13 by brianmcgee