nix-community/nixos-facter #79: Compare cpu entries in hardware with /proc/cpuinfo

brianmcgee, 2024-08-14:
Ensure we are capturing comparable information in our report.

brianmcgee, 2024-08-16:
`hwinfo` is parsing `/proc/cpuinfo` already, but there are a few 'newer' fields which should be added, such as:

- `bugs`
- `TLB size`
- `clflush size`
- `cache_alignment`
- `address sizes`
- `power management`
- `microcode`
- `cpu cores`
- `core id`
- `apicid`
- `initial apicid`
- `fpu`
- `fpu_exception`
- `cpuid level`

I think, for the time being, we are capturing the most important and relevant attributes across architectures. If we hit a point writing NixOS modules and find ourselves needing one of the above, we could address the differences then.

Thoughts @zimbatm @Mic92 ?

Mic92, 2024-08-16:
I think core id / cpu cores is the most important one, I would start with these if you attempt to make a pull request and add the other ones later.

closed 2024-08-16 by brianmcgee