nix-community/nixos-facter #125: failed to scan: failed to add to smbios report: language field is already set

rcambrj, 2024-10-13:
**Describe the bug**

```
failed to scan: failed to add to smbios report: language field is already set
````

**To Reproduce**

```
sudo nixos-facter --hardware bios
```

**Expected behavior**

nixos-facter generates a valid facter.json

**System information**

Running on a Topton 4-port x86_64 Celeron N5105.
[image]

**Additional context**

The error seems to come from [this line](https://github.com/numtide/nixos-facter/blob/185a2d18eea4749d19c7abc4aff770354bb19446/pkg/facter/smbios.go#L118). The output of `dmidecode` seems to indicate that indeed, the language is supplied twice. I don't know what this really means, but hopefully it helps.

```
sudo dmidecode -t bios
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: American Megatrends International, LLC.
	Version: 5.19
	Release Date: 04/24/2022
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 8 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		BIOS ROM is socketed
		EDD is supported
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 5.19

Handle 0x0017, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 1
		en|US|iso8859-1
	Currently Installed Language: en|US|iso8859-1

Handle 0x005E, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Abbreviated
	Installable Languages: 1
		enUS
	Currently Installed Language: enUS
```

brianmcgee, 2024-10-13:
It wasn't 100% clear if `language` was a scalar or a vector. In the end we decided to make it scalar and assert so we would know if someone encountered this in the wild. 

You just confirmed it should be a vector 🙃🙂

rcambrj, 2024-10-13:
Thanks for the quick response.
Woohoo! I'm very happy to have been the guinea pig 🥳

brianmcgee, 2024-10-13:
I can have a look at this tomorrow, it's a straightforward change. 

I highly doubt anyone is relying on this field in their nix config yet, so we can probably just push it out.

@Mic92 do you agree?

rcambrj, 2024-10-13:
No rush! `all-hardware.nix` fills the gap in the meantime.

closed 2024-10-26 by brianmcgee

hasskvist, 2025-04-12:
I have two "Chassis Information" which causes issue ```failed to scan: failed to add to smbios report: chassis field is already set```

```
sudo dmidecode
Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Dell Inc.
	Type: Detachable
	Lock: Not Present
	Version: Not Specified
	Serial Number: C6J0WT2
	Asset Tag: Not Specified
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Detachable

Handle 0x0300, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Dell Inc.
	Type: Docking Station
	Lock: Not Present
	Version: Not Specified
	Serial Number: C6J0WT2
	Asset Tag: Not Specified
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x000000FF
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Docking Station
```

commit abde94ea (2024-10-14): feat: change language in smbios to an array

When running nixos-facter on a Topton 4-port x86_64 Celeron N5105, the SMBIOS is returning multiple language entries.

```console
sudo dmidecode -t bios
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

...

Handle 0x0017, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 1
		en|US|iso8859-1
	Currently Installed Language: en|US|iso8859-1

Handle 0x005E, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Abbreviated
	Installable Languages: 1
		enU [... cut]
PR #127 (2024-10-26): fix: change language in smbios to an array
When running nixos-facter on a Topton 4-port x86_64 Celeron N5105, the SMBIOS is returning multiple language entries.

```console
sudo dmidecode -t bios
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

...

Handle 0x0017, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 1
		en|US|iso8859-1
	Currently Installed Language: en|US|iso8859-1

Handle 0x005E, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Abbreviated
	Installable Languages: 1
		enUS
 [... cut]
commit a7746033 (2025-04-12): fix: chassis can have multiple entries in smbios

Closes #125