nix-community/nixos-facter #133: smbios missing in report for Raspberry Pi 4

pizzapim, 2024-11-03:
**Describe the bug**

The `smbios` part of the report is missing for my Raspberry Pi 4 system. This makes it hard for me to check whether the machine is a Raspberry Pi. I don't think `smbios` is available because `dmidecode` also reports an error. Perhaps `lshw` can be used to similar data?

Full report:
[facter.json](https://github.com/user-attachments/files/17612009/facter.json)

brianmcgee, 2024-11-04:
@pizzapim what do you mean when you say that `lshw` can be used to get similar data? I don't have access to a Pi 4 to check it's output. Could you provide a dump?

Mic92, 2024-11-04:
We can just read it for from device-tree:

```
[root@matchbox:~]# cat /proc/device-tree/model
Pine64 Rock64
```

pizzapim, 2024-11-04:
Pretty sure `lshw` does that under the hood as well.

Mic92, 2024-11-04:
Yes. This is where I got the idea from.