nix-community/nixos-facter #136: Put all "unknown" devices that are connected via USB into usb.

Mic92, 2024-11-04:
That's more intuitive: 

https://github.com/Mic92/dotfiles/blob/b8db102dab16013d48af91bb674ccd1160444e0f/machines/turingmachine/facter.json#L3929

brianmcgee, 2024-11-04:
I can have a look later this week

Mic92, 2024-11-05:
We should probably do the same for PCI devices.

brianmcgee, 2024-11-05:
Ok, it looks like the hardware class is a composite field which can encode multiple hardware classes. In the textual output of hwinfo there's a `HW Class List` 

```
69: USB 00.1: 0000 Unclassified device
  [Created at usb.122]
  Unique ID: JRd1.nXPV_d5LjE5
  Old Unique ID: FrNE.6cEYkEA3jA6
  Parent ID: Uc5H.lRoCB54l1cE
  SysFS ID: /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-4/1-4.3/1-4.3:1.1
  SysFS BusID: 1-4.3:1.1
  Hardware Class: unknown
  HW Class List: usb
  Model: "Yubico.com Yubikey 4/5 OTP+U2F+CCID"
  Hotplug: USB
  Vendor: usb 0x1050 "Yubico.com"
  Device: usb 0x0407 "Yubikey 4/5 OTP+U2F+CCID"
  Revision: "5.43"
  Driver: "usbhid"
  Driver Modules: "usbhid"
  Main Driver Module: "usbhid"
  Speed: 12 Mbps
  Module Alias: "usb:v1050p0407d0543dc00dsc00dp00ic03isc00ip00in01"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #87 (Hub)
```

In light of this, I need to review how we interpret hw class and also review how I made changes to it for webcam support.

PR #146 (2024-11-05): fix: classification of pci and usb devices
Improves overall classification by capturing the class list of the underlying hardware item.

When the overall hardware class is unknown, we call back to the class list which typically identifies it as usb or pci etc.

Adds the `class` and `class_list` fields to the JSON output for clarity.

We should consider whether we allow the same hardware item in different sections of the hardware report based on the `class_list`.

Closes #136
closed 2024-11-05 by brianmcgee