nix-community/nixos-facter #140: nixos-facter is missing hardware that is present in hwinfo

Mic92, 2024-11-05:
In `hwinfo` I see this:

46: USB 00.0: 0000 Unclassified device
  [Created at usb.122]
  Unique ID: POWV.ICYKfTdMza8
  Parent ID: k4bc.2DFUsyrieMD
  SysFS ID: /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0
  SysFS BusID: 1-9:1.0
  Hardware Class: unknown
  Model: "Synaptics Prometheus MIS Touch Fingerprint Reader"
  Hotplug: USB
  Vendor: usb 0x06cb "Synaptics, Inc."
  Device: usb 0x00bd "Prometheus MIS Touch Fingerprint Reader"
  Serial ID: "9fb43bdff866"
  Speed: 12 Mbps
  Module Alias: "usb:v06CBp00BDd0000dcFFdsc10dpFFicFFisc00ip00in00"
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #52 (Hub)

However in my factor.json the same device is not present:

[facter.json](https://github.com/user-attachments/files/17628535/facter.json)

Mic92, 2024-11-05:
Looking at the facter.json I am missing devices between 42 and 47?

Mic92, 2024-11-05:
I added some logging and it looks like this device is also received in factor but than discarded?

118 sysfs_id: /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0
119 sysfs_bus_id: 1-9:1.0
120 sysfs_device_link:

Mic92, 2024-11-05:
Ah actually it's in the report:

```
      {
        "index": 40,
        "attached_to": 46,
        "bus_type": {
          "name": "USB",
          "value": 134
        },
        "slot": {
          "bus": 0,
          "number": 0
        },
        "base_class": {
          "name": "Unclassified device",
          "value": 0
        },
        "sub_class": {
          "name": "Unclassified device",
          "value": 0
        },
        "vendor": {
          "value": 1739
        },
        "device": {
          "value": 189
        },
        "serial": "9fb43bdff866",
        "model": "Unclassified device",
        "sysfs_id": "/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0",
        "sysfs_bus_id": "1-9:1.0",
        "resources": [
          {
            "type": "baud",
            "speed": 12000000,
            "bits": 0,
            "stop_bits": 0,
            "parity": 0,
            "handshake": 0
          }
        ],
        "detail": {
          "device_class": {
            "name": "vendor_spec",
            "value": 255
          },
          "device_subclass": {
            "name": "audio_video",
            "value": 16
          },
          "device_protocol": 255,
          "interface_class": {
            "name": "vendor_spec",
            "value": 255
          },
          "interface_subclass": {
            "name": "per_interface",
            "value": 0
          },
          "interface_protocol": 0,
          "interface_number": 0,
          "interface_alternate_setting": 0
        },
        "hotplug": "usb",
        "module_alias": "usb:v06CBp00BDd0000dcFFdsc10dpFFicFFisc00ip00in00"
      }
    ],
```

But for some reasons we are missing the name?

brianmcgee, 2024-11-05:
The `nixpkgs` input for `numtide/hwinfo` was from August. You added the change to source latest USB and PCI ids from systemd in September. I think you're seeing an out of date set of reference data.

I think #147 fixes this. Give it a try.

Mic92, 2024-11-05:
I actually have to test it on the laptop that had the issue. But I can only do this later.

brianmcgee, 2025-01-07:
@Mic92 did you retest, it was still an issue and you re-opened this?

Mic92, 2025-01-08:
Works: https://github.com/Mic92/dotfiles/pull/1434

commit 5cda4951 (2024-11-05): feat: update hwinfo input

Brings in latest usb and pci ids from systemd.

Closes #140
PR #147 (2024-11-05): feat: update hwinfo input
Brings in latest usb and pci ids from systemd.

Closes #140
commit 50199460 (2024-11-05): feat: update hwinfo input

Brings in latest usb and pci ids from systemd.

Closes #140
closed 2025-01-08 by Mic92