nix-community/nixos-facter #171: `captureTouchpads` segfaults

guiniol, 2025-01-03:
**Describe the bug**

facter panics in `captureTouchpads`. This happens both at v0.3.0 and HEAD. The trace below is from HEAD:

```
2025/01/03 21:18:29 INFO building report system=x86_64-linux version=1
2025/01/03 21:18:29 INFO scanning hardware features="[memory pci net serial cpu bios monitor scsi usb prom sbus sys sysfs udev block wlan]"
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0xa pc=0x520f86]

goroutine 1 [running]:
github.com/numtide/nixos-facter/pkg/hwinfo.captureTouchpads(0x4a)
        github.com/numtide/nixos-facter/pkg/hwinfo/input.go:21 +0xc6
github.com/numtide/nixos-facter/pkg/hwinfo.Scan({0xc000098080, 0x10, 0x6f5780?})
        github.com/numtide/nixos-facter/pkg/hwinfo/hwinfo.go:82 +0x59c
github.com/numtide/nixos-facter/pkg/facter.(*Scanner).Scan(0x6d3c00)
        github.com/numtide/nixos-facter/pkg/facter/facter.go:76 +0x273
github.com/numtide/nixos-facter/cmd.Execute()
        github.com/numtide/nixos-facter/cmd/root.go:118 +0x2fd
main.main()
        github.com/numtide/nixos-facter/main.go:8 +0xf
```

**To Reproduce**

Steps to reproduce the behavior:

1. `nix run --option experimental-features "nix-command flakes" #nixos-facter -- -o facter.json`

**Expected behavior**

`nixos-facter` produces `facter.json` without segfaulting.

**System information**

Instead here are the list of parts, and I am also attaching a `facter.json` generated with v0.2.0.
[facter.json](https://github.com/user-attachments/files/18303974/facter.json)

- Asrock B550m Pro 4
- Ryzen 5 Pro 5650GE
- 2x Kingston KSM32ED8/32HC 32GB
- Kioxia Exceria G2 1TB (boot drive)
- 6x PM863A 3.84TB
- Connectx4-Lx 2\*25Gb with a single 10G fiber transceiver
- HDPLEX 250W GaN Passive
- Streacom FC10 ST-FC10B-ALPHA

I `git bisect`-ed it down to b7473ddde90e0c7dd180349a52ee961044ce4cbc, which doesn't really help since it's the commit that adds `input.go`.
I thought maybe it would help if it had a perip

brianmcgee, 2025-01-04:
Thanks for the report. I should have some time next week to look into this.

brianmcgee, 2025-01-08:
I believe this will be resolved by #172 . I changed how I read udev data and it should account for the nil pointer happening here.

brianmcgee, 2025-01-09:
@guiniol I'll add some logging tomorrow and then have you run it again to work out what's going on. 

I see it's failing after pulling udev data. I'll examine the underlying code in hwinfo to see if there's any obvious edge cases.

guiniol, 2025-01-09:
Thanks! Happy to help however I can.

brianmcgee, 2025-01-10:
@guiniol please try #174

guiniol, 2025-01-11:
It works!!

```
2025/01/11 22:56:06 INFO building report system=x86_64-linux version=1
2025/01/11 22:56:06 INFO scanning hardware features="[memory pci net serial cpu bios monitor scsi usb prom sbus sys sysfs udev block wlan]"
2025/01/11 22:56:07 INFO reading IOMMU groups
2025/01/11 22:56:07 INFO processing devices count=67
2025/01/11 22:56:07 INFO processing smbios entries count=27
2025/01/11 22:56:07 INFO detecting virtualisation
2025/01/11 22:56:07 INFO report complete
```

(and sorry for the delay, couldn't make it to the computer yesterday)

guiniol, 2025-01-11:
Ran it again with `--log-level debug` and I see the log line you add in #174:

```
2025/01/11 23:12:52 input.go:35: DEBUG udev data missing input name="ASRock LED Controller" sysfs=/devices/pci0000:00/0000:00:02.1/0000:02:00.0/usb1/1-8/1-8:1.0/0003:26CE:01A2.0002/input/input2
```

I didn't even know I had a LED controller, but that's what was causing the issue :smile:

Mic92, 2025-01-14:
Can be closed than.

closed 2025-01-14 by Mic92

guiniol, 2025-01-14:
Awesome. Thanks again!

PR #172 (2025-01-08): feat: use linux input Bus enum with udev data
I'm not sure why I created a separate Bus enum for udev. From the source, as best I can tell, it's using https://github.com/systemd/systemd/blob/8e3347f3bd3d9a01b8f39b0858eab74084ecf20a/src/basic/linux/input.h#L252-L275.

This aligns with that and should fix #166.

Closes #166 
Closes #171