nix-community/nixos-facter #183: A facter.json diff

mightyiam, 2025-02-22:
I may have replaced a USB mouse, but otherwise I don't think these changes are desirable.

[facter.json.txt](https://github.com/user-attachments/files/18921736/facter.json.txt)

Sorry for dumping all this here in one issue, but that's all I'm willing to do at the moment.

mightyiam, 2025-03-14:
Here is another, where the only significant event that occurred in between is a reboot:

[facter.json.diff.txt](https://github.com/user-attachments/files/19238855/facter.json.diff.txt)

Looks like much of the differences can be solved by sorting. Some others by filtering.

brianmcgee, 2025-07-07:
Most of the diffs seem formatting related? I don't remember making any changes to the `json` output that would cause this.

mightyiam, 2025-07-08:
Sorry, that is contaminated with formatting. In any case, same general issue as #207. Closing in favor of that.

closed 2025-07-08 by mightyiam