nix-community/nixos-facter #184: facer doesn't like my pci devices

marcusramberg, 2025-03-19:
```
❯ sudo nix run \
  --option experimental-features "nix-command flakes" \
  --option extra-substituters https://numtide.cachix.org \
  --option extra-trusted-public-keys numtide.cachix.org-1:2ps1kLBUWjxIneOy1Ik6cQjb41X0iXVXeHigGmycPPE= \
  github:numtide/nixos-facter -- -o facter.json
/nix/store/p6k7xp1lsfmbdd731mlglrdj2d66mr82-bash-5.2p37/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8): No such file or directory
2025/03/19 20:48:13 INFO building report system=x86_64-linux version=1
2025/03/19 20:48:13 INFO scanning hardware features="[memory pci net serial cpu bios monitor scsi usb prom sbus sys sysfs udev block wlan]"
2025/03/19 20:48:14 failed to scan: failed to scan hardware: failed to fetch udev data for device "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:0557:2221.0001/input/input4" with udev data: failed to parse usb: failed to parse model id: strconv.ParseUint: parsing "": invalid syntax
❯
```

This is from a  Supermicro X9SRE/X9SRE-3F/X9SRi/X9SRi-3F 

```
 cat /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:0557:2221.0001/input/input4/name -p
Winbond Electronics Corp Hermon USB hidmouse Device
```

I guess this is a mouse device provided by ipmi.

brianmcgee, 2025-03-20:
Would you mind sharing the output of `cat /run/udev/data/+input:input4`?

wantguns, 2025-03-23:
Hi, I am also facing the same issue:
```
wantguns@bellatrix:~$ sudo nix run nixpkgs#nixos-facter -- -o facter.json
2025/03/23 12:47:06 INFO building report system=aarch64-linux version=1
2025/03/23 12:47:06 INFO scanning hardware features="[memory pci net serial cpu bios monitor scsi usb prom sbus sys sysfs udev block wlan]"
2025/03/23 12:47:06 failed to scan: failed to scan hardware: failed to fetch udev data for device "/sys/devices/pci0000:00/0000:00:02.0/usb1/1-1/1-1:1.0/0003:0627:0001.0001/input/input1" with udev data: failed to parse usb: failed to parse model id: strconv.ParseUint: parsing "": invalid syntax

wantguns@bellatrix:~$ cat /run/udev/data/+input:input1
I:2578106
E:ID_INPUT=1
E:ID_INPUT_MOUSE=1
E:ID_VENDOR=QEMU
E:ID_VENDOR_ENC=QEMU
E:ID_VENDOR_ID=0627
E:ID_MODEL=QEMU_USB_Tablet
E:ID_MODEL_ENC=QEMU\x20USB\x20Tablet
E:ID_MODEL_ID=0001
E:ID_REVISION=0000
E:ID_SERIAL=QEMU_QEMU_USB_Tablet_28754-0000:00:02.0-1
E:ID_SERIAL_SHORT=28754-0000:00:02.0-1
E:ID_TYPE=hid
E:ID_BUS=usb
E:ID_USB_INTERFACES=:030000:
E:ID_USB_INTERFACE_NUM=00
E:ID_USB_DRIVER=usbhid
E:ID_PATH=pci-0000:00:02.0-usb-0:1:1.0
E:ID_PATH_TAG=pci-0000_00_02_0-usb-0_1_1_0
E:ID_FOR_SEAT=input-pci-0000_00_02_0-usb-0_1_1_0
G:seat
```

I did some debugging and found out that the `ID_USB_MODEL_ID` was added to udev in [systemd v257.4](https://github.com/systemd/systemd/commit/479da1107a0d4e2f7ef5cd938512b87a0e45f180) and I am running an older version.

Fixed the error by adding the udev env manually, by first adding this udev rule:
```
# inside /etc/udev/rules.d/99-usb-model.rules

# Add USB prefix properties to all USB devices
SUBSYSTEM=="usb|input", ENV{ID_BUS}=="usb", ENV{ID_MODEL_ID}=="?*", ENV{ID_USB_MODEL_ID}!="?*", ENV{ID_USB_MODEL_ID}="$env{ID_MODEL_ID}"
SUBSYSTEM=="usb|input", ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="?*", ENV{ID_USB_VENDOR_ID}!="?*", ENV{ID_USB_VENDOR_ID}="$env{ID_VENDOR_ID}"
SUBSYSTEM=="usb|input", ENV{ID_BUS}=="usb", ENV{ID_REVISION}=="?*", ENV{ID_USB_REVISION}!="?*", ENV{ID_USB_REVISION}="$env{ID_REVISION}"
```

and after reloading and trigger the subsystems:
```bash
sudo udevadm control --reload-rules
sudo udevadm trigger --action=add --subsystem-match=usb
sudo udevadm trigger --action=add --subsystem-match=input
```

I get the env populated:
```
wantguns@bellatrix:~$ cat /run/udev/data/+input:input1 | grep ID_USB
E:ID_USB_INTERFACES=:030000:
E:ID_USB_INTERFACE_NUM=00
E:ID_USB_DRIVER=usbhid
E:ID_USB_MODEL_ID=0001
E:ID_USB_VENDOR_ID=0627
E:ID_USB_REVISION=0000
```

brianmcgee, 2025-03-24:
@wantguns thanks for the detailed breakdown.

I'll add a check for the systemd version.

marcusramberg, 2025-03-25:
Nice. I was doing this on an ancient arch machine, which I've now reinstalled with nix, so I'm guessing systemd version was my issue as well.

commit ad96a02e (2025-04-08): feat: add a check for a minimal udev version

Closes #184
PR #194 (2025-04-08): feat: add a check for a minimal udev version
Closes #184
commit 3b189913 (2025-04-08): feat: add a check for a minimal udev version

Closes #184

Update pkg/udev/udev.go

Update pkg/udev/udev.go
closed 2025-04-08 by Mic92