nix-community/nixos-facter #185: uint64 can exceed Nix integer range

Scandiravian, 2025-03-28:
An integer in Nix is always signed, which can lead to issues when a number exceeds `2^63 - 1`.

For instance, I ran nixos-facter on a Dell R760 server, which sets the value of an entry in `hardware.memory_array_mapped_address` to `18446741876833779712`. This makes the nix configuration fail to build when anything references the report as JSON seems to be eagerly loaded by nix.

This is clearly an edge case, but would it be possible to use strings instead of uint64 in the output for these values?

brianmcgee, 2025-03-31:
🤔 I'll have a look a bit later this week.

Scandiravian, 2025-04-04:
Please let me know if there's anything I can do to help. I have some experience in Go and since this related to my work, I can spend a bit of time on it.

Do you know what an ideal solution to this would look like for you (if you think this should be fixed)? If you let me know I can look into implementing it

Thanks for all the work you're putting into this project - facter is an extremely useful tool for me!

brianmcgee, 2025-04-04:
@Scandiravian, I'm glad to hear you find it helpful. 

I will try to think through the implications for a bit today. I'll post my thoughts here. 

Happy to review PRs if you want to implement the fix 🙏

songpola, 2025-04-06:
Yes. I got this error too: `error: unsigned json number 9223372036854775808 outside of Nix integer range`

Cause by:
```shell
$ jq '.smbios.memory_64_error[] | {array_address, device_address}' facter.json
{
  "array_address": 9223372036854775808,
  "device_address": 9223372036854775808
}
```

brianmcgee, 2025-04-07:
A quick review of mapping from `hwinfo` s underlying data model shows some places I can reduce the size from `uint`. 

I'll spend some time today doing this and converting any explicit usages of `uint64_t` to `string` and see if anything breaks in `nixos-facter-modules`. I don't think any values are actively being used, so I can safely break the format without a version bump.

Mic92, 2025-04-07:
@brianmcgee maybe a those addresses could be formatted as hex strings. That's also more readable.

Scandiravian, 2025-04-14:
I've just tested #193 on the R760 servers, that were experiencing the issue and the fix resolved it 😊

closed 2025-04-21 by Mic92