nix-community/nixos-facter #186: CUDA support detection

yajo, 2025-03-29:
A system with an NVIDIA GPU card with CUDA capabilities shows this:

```shell
$ nvidia-settings -q CUDACores -t
2048
```

You can also get the CUDA version with `nvidia-smi -q`.

It would be nice to include that in reports. For example, you could [enable CUDA automatically](https://nixos.org/manual/nixpkgs/stable/#opt-cudaSupport) (or behind some option that allows facter enabling stuff that will cause mass rebuilds).

@moduon MT-9339

yajo, 2025-07-02:
> the report should just give the most simple base information

Isn't it the opposite? The report should be as detailed as possible. We want modules that react to facts. So, the more facts we gather, the better, right?

Regarding the scope of the modules, I think that discussion belongs to https://github.com/nix-community/nixos-facter-modules/issues/72. In this repo we're only talking about gathering information, not about what to do with it.

brianmcgee, 2025-07-07:
The approach is to gather as much interesting information as we can about a machine. We only remove info from the report if it could be considered sensitive or if it is likely to change between restarts (#207). 

We want to allow module authors as much flexibility as possible later on, whilst presuming as little as possible. .

clotodex, 2025-07-07:
Makes sense.
How does it work with "dynamic" information then? 
I.e. is it possible to discover cuda capabilities statically? Or is that itself dependent on the newest cuda versions and their compatibility with your card. This is the thing I meant - I am unsure - specifically with nvidia - what is static, dynamic, should be inferred, or impossible to gather. I don't mind much either way, just tried to understand where what goes :)

yajo, 2025-09-25:
To help you understand:

- static: the JSON file contents.
- dynamic: the tool that generates the JSON file.