nix-community/nixos-facter #203: panic: runtime error: invalid memory address or nil pointer dereference

Luflosi, 2025-05-09:
```
❯ sudo nixos-facter --log-level debug -o facter.json 
2025/05/09 12:13:17 facter.go:67: DEBUG building report system=x86_64-linux version=1
2025/05/09 12:13:17 facter.go:74: DEBUG scanning hardware features="[memory pci net serial cpu bios monitor scsi usb prom sbus sys sysfs udev block wlan]"
2025/05/09 12:13:17 facter.go:81: DEBUG reading IOMMU groups
2025/05/09 12:13:17 facter.go:89: DEBUG processing devices count=98
2025/05/09 12:13:17 hardware.go:190: DEBUG hardware.add index=1 attached_to=0 class=bios class_list="[bios vesa_bios]" bus=<nil> sysfs_id=""
2025/05/09 12:13:17 hardware.go:190: DEBUG hardware.add index=2 attached_to=0 class=system class_list=[system] bus=<nil> sysfs_id=""
2025/05/09 12:13:17 hardware.go:190: DEBUG hardware.add index=3 attached_to=0 class=cpu class_list=[cpu] bus=<nil> sysfs_id=""
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0xc0 pc=0x5453a7]

goroutine 1 [running]:
github.com/numtide/nixos-facter/pkg/facter.(*Hardware).add.func1(0xc00027cc60?, 0xc000037718?)
        github.com/numtide/nixos-facter/pkg/facter/hardware.go:255 +0x7
slices.insertionSortCmpFunc[...](...)
        slices/zsortanyfunc.go:12
slices.pdqsortCmpFunc[...]({0xc00027cc60, 0x2, 0x2}, 0x56fa20, 0xc?, 0xc000037818?, 0x5af980?)
        slices/zsortanyfunc.go:73 +0x48f
slices.SortFunc[...](...)
        slices/sort.go:32
github.com/numtide/nixos-facter/pkg/facter.(*Hardware).add(_, {0x3, 0x0, 0x2, {0xc000195a60, 0x1, 0x1}, 0x0, 0x0, 0xc00020ce40, ...})
        github.com/numtide/nixos-facter/pkg/facter/hardware.go:254 +0x685
github.com/numtide/nixos-facter/pkg/facter.(*Scanner).Scan(0x6da6c0)
        github.com/numtide/nixos-facter/pkg/facter/facter.go:101 +0x658
github.com/numtide/nixos-facter/cmd.Execute()
        github.com/numtide/nixos-facter/cmd/root.go:118 +0x2fd
main.main()
        github.com/numtide/nixos-facter/main.go:8 +0xf
```

Using nixos-facter version 0.3.2.

The CPU is an old Intel Xeon CPU (E5606).
Please let me know if you need any additional information, such as the output of `/proc/cpuinfo`.

brianmcgee, 2025-05-12:
Looks like we're not getting a `physical id` field being reported for your CPU, so yeah, can you provide the output of `/proc/cpuinfo`.

Luflosi, 2025-05-12:
Thanks for taking a look! Here it is:
<details>
<summary>cat /proc/cpuinfo</summary>

```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 44
model name	: Intel(R) Xeon(R) CPU           E5606  @ 2.13GHz
stepping	: 2
microcode	: 0x1f
cpu MHz		: 2127.852
cache size	: 8192 KB
physical id	: 1
siblings	: 4
core id		: 0
cpu cores	: 4
apicid		: 32
initial apicid	: 32
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 popcnt aes lahf_lm pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid dtherm arat vnmi flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown
bogomips	: 4255.72
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 44
model name	: Intel(R) Xeon(R) CPU           E5606  @ 2.13GHz
stepping	: 2
microcode	: 0x1f
cpu MHz		: 2127.864
cache size	: 8192 KB
physical id	: 1
siblings	: 4
core id		: 1
cpu cores	: 4
apicid		: 34
initial apicid	: 34
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 popcnt aes lahf_lm pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid dtherm arat vnmi flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown
bogomips	: 4255.72
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 6
model		: 44
model name	: Intel(R) Xeon(R) CPU           E5606  @ 2.13GHz
stepping	: 2
microcode	: 0x1f
cpu MHz		: 2127.862
cache size	: 8192 KB
physical id	: 1
siblings	: 4
core id		: 9
cpu cores	: 4
apicid		: 50
initial apicid	: 50
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 popcnt aes lahf_lm pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid dtherm arat vnmi flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown
bogomips	: 4255.72
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management:

processor	: 3
vendor_id	: GenuineIntel
cpu family	: 6
model		: 44
model name	: Intel(R) Xeon(R) CPU           E5606  @ 2.13GHz
stepping	: 2
microcode	: 0x1f
cpu MHz		: 2127.879
cache size	: 8192 KB
physical id	: 1
siblings	: 4
core id		: 10
cpu cores	: 4
apicid		: 52
initial apicid	: 52
fpu		: yes
fpu_exception	: yes
cpuid level	: 11
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid dca sse4_1 sse4_2 popcnt aes lahf_lm pti ssbd ibrs ibpb stibp tpr_shadow flexpriority ept vpid dtherm arat vnmi flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only ept_1gb flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest ple
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown
bogomips	: 4255.72
clflush size	: 64
cache_alignment	: 64
address sizes	: 40 bits physical, 48 bits virtual
power management:
```
</details>

brianmcgee, 2025-05-29:
Hmm, I see `physical id` in both outputs above. 

This fell off my radar a bit, I'll try to have a look in the next few days.

brianmcgee, 2025-06-02:
I've created a fix #204.

Can you give it a try with `sudo nix run github:numtide/nixos-facter/fix/parsing-cpu-info -- -o report.json`?

Luflosi, 2025-06-03:
I can confirm that this fixes the crash.

Thank you!